microhal/hardwaretester #6: MCP23017 device PDB design.

sakoPO, 2018-03-19:


pjacalski, 2018-03-23:
Initial commit for PCB design.
https://circuitmaker.com/projects/details/Pawel-Jacalski/UHALMCP9800#sectionDesignFiles

sakoPO, 2018-03-26:
Please add pin headers too all gpio. To perform test we should:
- connect at least one pin to VCC by 270R resistor
- connect at least one pin to GND by 270R resistor.
- connect two different pins each other by 270R resistor.
when this resistors will be removed board should work as standard expander.
We should consider adding LED to one of GPIOs.

pjacalski, 2018-03-29:
Added updates.
https://workspace.circuitmaker.com/Projects/Details/Pawel-Jacalski/UHALMCP23017LOOPBACKVerB

sakoPO, 2018-04-11:
Please update resistors values in loopback ver b. Note from third point should using two pins that wasn't used previously.

pjacalski, 2018-04-12:
Fixed.

sakoPO, 2018-04-12:
If possible delete projects:
https://circuitmaker.com/projects/details/Pawel-Jacalski/UHALMCP23017
https://circuitmaker.com/projects/details/Pawel-Jacalski/UHALMCP23017LOOPBACK

after deleting project task can be closed.

pjacalski, 2018-04-13:
Done.

closed 2018-04-13 by pjacalski